wpilibsuite/RobotBuilder #32: Add XboxController

AustinShalit, 2017-01-18:


cpapplefamily, 2020-07-24:
If we change the line in the `RobotContainer-declarations.java` from
`"#type($component)" == "Joystick"`
to
`$component.getBase().getType() == "Joystick"`

and `RobotContainer-constructors.java` from
`"#type($component)" != "Joystick"`
to
`$component.getBase().getType() != "Joystick"`

We can open the ability to create extensions. I have tested this and created Xbox controller and Xbox Button extensions. Or we will need to add more conditions to the if statements to allow the $component type to validate the xboxController class. It seems the `#type($component)` returns the class not the type. `$component.getBase().getType()` seems to return the component type.

I suspect that the same if conditions exist and can be altered in the CPP exporters.

cpapplefamily, 2020-07-24:
One issue I did have when creating the extensions was I could not get the #button macro to create the parameters needed to add the GUI instances for the Xbox Button. I had to manualy add them all to the extension

cpapplefamily, 2020-10-23:
@AustinShalit and others I have created an extension to add the XBoxController. The issue I am having is the macro `#type($component)` that returns the type seems to return the Class Name not the type. If I can figure how to get the type not the class name from the $component variable changing the areas that reference `#type($component)` can sort the object correctly with extensions that have class name XboxController but are of the Joystick Type.

AustinShalit, 2020-10-23:
@sciencewhiz Is going to be the right person to ask about that.

cpapplefamily, 2020-10-23:
Thanks for the reply. Through brute force and investigative logic I found `$component.getBase().getType()` returned what I was looking for. with the Pull request #269 opens up the possibilities of other joystick types getting added to the OI section. We us the MSP430 launchPad and could maybe create a add on for that tool

PR #335 (2020-12-22): Add Xbox Controller Support
Supersedes #271 
Fixes #32
closed 2020-12-22 by sciencewhiz