wpilibsuite/RobotBuilder #34: Save project settings in home/Robotbuilder

SamCarlberg, 2017-01-19:
This would allow teams to have their RobotBuilder project files to be used by multiple people and let it be in their version control system.

Something like

```
~
|_ Robotbuilder
  |_ settings
    |_ gearsbot.yml
    |_ 2017bot.yml
    |_ practicebot.yml
    |_ ...
```

With the settings YAML files storing workspace, wiring file, etc.

```yaml
workspace: /home/sam/workspace
simulation_world_file: ...
wiring_file_location: ...
```

sciencewhiz, 2017-01-22:
We store the robotbuilder yaml file in the root of the exported project, using a relative path to the eclipse workspace of ../. This lets us store the yaml file in git, and not worry about absolute paths.

sciencewhiz, 2017-03-20:
It turns out that approach only works on windows. We just got a mac laptop setup, and it created a directory "../"