wpilibsuite/RobotBuilder #36: Robotbuilder fails to load existing yml file.

danielcastellarin, 2017-02-04:
We have one laptop using plugins version 2017.2.1 and one using 2017.1.1.  The 2.1 version fails to load yml save file.  The 1.1 works. When opening thru eclipse we get no output/result until we exit eclipse. Then exception/error window pops up.  If we open Robotbuilder directly same thing happens after choosing to load existing project.
[image]

Our Robotbuilder.yml file can be found here: https://github.com/BHSRobotix/Steamworks2017/blob/master/RobotBuilder.yaml

sciencewhiz, 2017-02-04:
Have you installed the CTRE toolsuite on the laptop that fails? Your yaml references CAN Talon, which is an extension installed by the CTRE toolsuite.

chriscostello9, 2017-02-04:
Yep. that was the problem(same team that opened this issue posting from different user/laptop). Thanks for the tip.

closed 2017-02-04 by SamCarlberg