wpilibsuite/RobotBuilder #38: RobotBuilder Hangs when generating C++ code

dravis123, 2017-02-13:
Hello,

I have been using RobotBuilder successfully on the same laptop (windows 7), but within the last week have been seeing a problem that is now impacting code generation. I'm seeing an issue where after I click "C++" to generate the code, the cursor stays as a swirling blue ring (indicates something is busy). At first I didn't think this had any impact but now I am not seeing code updated in the files when I try to generate code. Note I have probably generated code 50+ times previously on same laptop, didn't change any software. 

To test this, I have an empty command group and generate code. Then I add commands to the group and generate code, but no commands are added to the file.

I have refreshed the file, tried this with project closed, restarted eclipse and laptop. Now no code will generate.

-Ben Randolph

sciencewhiz, 2017-02-13:
If you delete the source file for the empty command, does it then generate correctly?

dravis123, 2017-02-14:
Yes, I was able to get the file to create when I deleted the other files. I can see that the command in question (ShootSingleBall) does not get the .cpp file updated, the timestamp is older. But the .cpp~, .h, and .h~ files do get new modified timestamps. Is this a file locking problem perhaps where RobotBuilder cannot get write access? It seems odd because this worked many many times in recent days.

dravis123, 2017-02-14:
I am going to try a new eclipse/WPI lib install and see if it works then.

dravis123, 2017-02-15:
I finally recreated this and found the issue. If you have a parameter preset starting with a number, the generate code hangs/fails with no error messages. In this case, I had "10ft" as a preset, and changed it to TenFt and the code generated fine.

Do you think it would be possible to add validation to these fields or at least an error message when generating code? Could prevent other people from banging their heads against the wall here. I am SW Dev so I don't mind chasing down bugs :-).

sciencewhiz, 2017-11-12:
It appears there is already code to handle this condition, I wonder why it didn't work. https://github.com/wpilibsuite/RobotBuilder/blob/c70bdcdffc8b03a2603949bc5846f9478348f3c8/src/main/java/robotbuilder/data/properties/ParameterDescriptor.java#L76

closed 2017-12-31 by bradamiller

dravis123, 2017-12-31:
Thanks for working on this! I'll be sure to put RBuilder through its paces next week... :-)

PR #82 (2017-12-31): Validate the name of a Parameter Set
Fixes #38
commit d5e6a543 (2017-12-31): Validate the name of a Parameter Set (#82)

Fixes #38