wpilibsuite/RobotBuilder #42: Comment error in Robot.java generated by RobotBuilder

AustinShalit, 2017-05-19:
_From @jee7s on May 19, 2017 3:20_

In the referenced file, there is a comment that reads, in part:

> If you change the name of this class or the package after creating this project, you must also update the manifest file in the resource directory.

We found the location of the package and class definitions to be in the project's build.properties file, not the manifest file. Could this comment be updated for clarity?

_Copied from original issue: wpilibsuite/allwpilib#541_

AustinShalit, 2017-05-19:
https://github.com/wpilibsuite/RobotBuilder/blob/master/src/main/resources/export/java/RobotTemplate.java#L12-L18

sciencewhiz, 2017-05-19:
This affects the eclipse plugins also.

PR #44 (2017-05-25): Fix comment about updating class or package
Fixes #42
commit db14d49a (2017-05-25): Fix comment about updating class or package (#44)

Fixes #42
closed 2017-05-25 by SamCarlberg