wpilibsuite/RobotBuilder #46: Adding a pointer to the TalonSRX resources.

cpapplefamily, 2017-09-26:
During set up of Robotbuilder we point to our Eclipse Workspace, Simulation File, Wiring file location and a few other customization. Would it be possible to Identify the TalonSRX Jar and equivalent C++ file locations and add that to the project if a SRX was selected as a motor controller? Now that third party devices are not included in the WPILIB but by the manufacture we need to add these to the project manually.  Although knowing how to do this in Eclipse is not a bad skill. We also have to do this with our NavX Micro that simply can't be added using Robotbuilder.

Kevin-OConnor, 2017-09-26:
You should not have to add these locations manually. If these files are installed to the "correct" location, the plugins should be adding them to the project automatically.

sciencewhiz, 2017-10-01:
Further, Talon SRX was removed from robot builder (https://github.com/wpilibsuite/RobotBuilder/commit/93e4baf98e9b2dbd830f97d59587c63affe70a2f) and CTRE implemented it as a robot builder plugin. I'm not sure how you can get a case where robot builder supports CAN Talon. but you don't have the library.

closed 2017-12-28 by Kevin-OConnor