wpilibsuite/RobotBuilder #49: Re-Exporting C++ with TimedCommand causes exception

sciencewhiz, 2017-10-07:
When re-exporting a project with a Timed Command, robot builder throws an exception when exporting the TimedCommand header.

```Saved type: TimedCommand
  New type: TimedCommand
Modifying C:\Users\Joe\workspace\TimedCommandCPP\src\Commands\TimedCommand1.cpp
Saved type: TimedCommand
  New type: TimedCommand
Modifying C:\Users\Joe\workspace\TimedCommandCPP\src\Commands\TimedCommand1.h
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
        at java.io.Reader.<init>(Reader.java:78)
        at java.io.InputStreamReader.<init>(InputStreamReader.java:72)
        at robotbuilder.exporters.GenericExporter.evalResource(GenericExporter.java:237)
        at robotbuilder.exporters.ExportFile.export(ExportFile.java:61)
        at robotbuilder.exporters.GenericExporter.export(GenericExporter.java:164)
        at robotbuilder.actions.ExporterAction.actionPerformed(ExporterAction.java:36)
        at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
        at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
        at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
        at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
        at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
        at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
        at java.awt.Component.processMouseEvent(Component.java:6533)
        at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
        at java.awt.Component.processEvent(Component.java:6298)
        at java.awt.Container.processEvent(Container.java:2236)
        at java.awt.Component.dispatchEventImpl(Component.java:4889)
        at java.awt.Container.dispatchEventImpl(Container.java:2294)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
        at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
        at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
        at java.awt.Container.dispatchEventImpl(Container.java:2280)
        at java.awt.Window.dispatchEventImpl(Window.java:2746)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.awt.EventQueue$4.run(EventQueue.java:729)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

This occurs with the attached yaml file, which is basically as simple as it can get.

Steps to reproduce:
1) Load [rb.yaml.txt](https://github.com/wpilibsuite/RobotBuilder/files/1364298/rb.yaml.txt)
2) Export to C++
3) Export to C++ again

sciencewhiz, 2017-10-12:
@SamCarlberg any thoughts? I can't figure it out.

SamCarlberg, 2017-10-12:
Caused by https://github.com/wpilibsuite/RobotBuilder/blob/master/src/main/resources/export/cpp/files.yaml#L148

Shouldn't this be `TimedCommand-constructor-header.h`?

sciencewhiz, 2017-10-12:
That looks like it, thanks

PR #52 (2017-10-13): fix export of TimedCommand
Fixes #49
commit 7f85812a (2017-10-13): fix export of TimedCommand (#52)

Fixes #49
closed 2017-10-13 by PeterJohnson