wpilibsuite/RobotBuilder #83: RobotBuilder generated projects that require 3rd part, require eclipse close-reopen

ozrien, 2017-12-28:
Originally reported in private CTRE testing....@JCaporuscio 

[1] Use robot builder to create a 3rd part MC, most likely Talon SRX or Victor SPX.
[2] Open project in eclipse (Java or C++)
[3] Import RB-genered project
[4] Project can't resolve namespaces/classes in third party [NOT OK]
[5] Close and reopen eclipse
[6] Project now resolves third party components [OK]

ozrien, 2017-12-28:
This might be what was going on in https://github.com/wpilibsuite/RobotBuilder/issues/46

Kevin-OConnor, 2017-12-28:
There are WPIlib hooks to update libs 
 on WPIlib Deploy, project creation and plugin load. The import process is generic and not handled by the WPIlib plugin. The only way we could do this would be to reimplement all of the Eclipse logic for 3rd party libs into RobotBuilder as well and set the paths on project export

Kevin-OConnor, 2017-12-28:
Joe Ross pointed out some spots this can be added in the docs short term. Maybe adding a button to the WPIlib menu in Eclipse to force a library refresh without deploy longer tern?

ozrien, 2017-12-28:
That's a nice low risk solution. My thought that if an RB plugin reports that it depends on third party, RB could tell user to restart eclipse after importing, or remind them to hit the future-refresh-button

sciencewhiz, 2018-07-02:
Closing as Eclipse won't be used in 2019

closed 2018-07-02 by sciencewhiz