wpilibsuite/RobotBuilder #84: Support for adapter classes

ozrien, 2017-12-28:
Latest version of TalonSRX/VictorSPX does not implement the WPILIB MC interfaces directly.  Instead the API has a getter for a WPILIB compliant object....
https://github.com/CrossTheRoadElec/Phoenix-frc-lib/blob/ab8286b370cb9cae60a0b5f0caf4cbb34402cc60/java/src/com/ctre/phoenix/motorcontrol/IMotorController.java#L196

However this makes supporting RB more difficult as Drivetrain construction requires an additional get call.

bradamiller, 2017-12-28:
With 5.0.8 RobotBuilder generates code that doesn't work because the TalonSRX class doesn't implement SpeedController. So in my case I have a robot with 4 motor drive and RobotBuilder creates a DifferentialDrive with two SpeedController groups (left and right side). The SpeedControllerGroup needs SpeedControllers as the arguments, same as DifferentialDrive if I only had 2 motors. Is there a way of just getting TalonSRX and VictorSPX to directly implement SpeedController?

sciencewhiz, 2017-12-28:
Is there a disadvantage to always calling getWPILIB_SpeedController() in Robot Builder generated code?

bradamiller, 2017-12-28:
Probably not, it just seems cleaner. I was curious why it didn't just implement the class instead of returning the proxy class. But I just tried the suggested RobotBuilder change and it seems to work except for a few typos in the posted solution.

sciencewhiz, 2018-01-13:
Is this OBE with the latest changes in Phoenix?

ozrien, 2018-01-14:
Since Phoenix now has compatible classes WPI_TalonSRX/WPI_VictorSPX, this is no longer an issue.

closed 2018-01-14 by ozrien