wpilibsuite/RobotBuilder #107: Build fails with JDK 11

sciencewhiz, 2018-09-22:
When using JDK 11, build fails with the following errors. The cause appears to be described here: https://bugs.openjdk.java.net/browse/JDK-8186648 and robot builder needs to be updated.

`C:\Users\Joe\Documents\GitHub\RobotBuilder\src\main\java\robotbuilder\data\RobotComponent.java:160: error: incompatible types: Vector<TreeNode> cannot be converted to Vector<RobotComponent>
            return children;
                   ^
C:\Users\Joe\Documents\GitHub\RobotBuilder\src\main\java\robotbuilder\data\RobotComponent.java:167: error: incompatible types: Vector<DefaultMutableTreeNode> cannot be converted to Vector<TreeNode>
        this.children = children;
                        ^
C:\Users\Joe\Documents\GitHub\RobotBuilder\src\main\java\robotbuilder\ConstantsAdderDialog.java:90: error: incompatible types: Vector<Vector> cannot be converted to Vector<Vector<Object>>
        Vector<Vector<Object>> dataVector = getTableModel().getDataVector();
                                                                         ^
C:\Users\Joe\Documents\GitHub\RobotBuilder\src\main\java\robotbuilder\graph\CommandCellEditorDialog.java:71: error: incompatible types: Vector<Vector> cannot be converted to Vector<Vector<Object>>
        Vector<Vector<Object>> dataVector = parameterTable.getModel().getDataVector();
                                                                                   ^
C:\Users\Joe\Documents\GitHub\RobotBuilder\src\main\java\robotbuilder\ParameterAdderDialog.java:88: error: incompatible types: Vector<Vector> cannot be converted to Vector<Vector<Object>>
        Vector<Vector<Object>> dataVector = getTableModel().getDataVector();
                                                                         ^
C:\Users\Joe\Documents\GitHub\RobotBuilder\src\main\java\robotbuilder\ParameterEditorDialog.java:98: error: incompatible types: Vector<Vector> cannot be converted to Vector<Vector<Object>>
        Vector<Vector<Object>> dataVector = getTableModel().getDataVector();
                                                                         ^
C:\Users\Joe\Documents\GitHub\RobotBuilder\src\main\java\robotbuilder\ParameterSetsEditorDialog.java:46: error: incompatible types: Vector<Vector> cannot be converted to Vector<Vector<Object>>
        Vector<Vector<Object>> tableData = setsTable.getModel().getDataVector();
                                                                             ^
C:\Users\Joe\Documents\GitHub\RobotBuilder\src\main\java\robotbuilder\ParameterSetsTable.java:306: error: getDataVector() in PresetsTableModel cannot override getDataVector() in DefaultTableModel
        public Vector<Vector<Object>> getDataVector() {
                                      ^
  return type Vector<Vector<Object>> is not compatible with Vector<Vector>
C:\Users\Joe\Documents\GitHub\RobotBuilder\src\main\java\robotbuilder\ParameterSetsTable.java:305: error: method does not override or implement a method from a supertype
        @Override
        ^
C:\Users\Joe\Documents\GitHub\RobotBuilder\src\main\java\robotbuilder\ParameterSetsTable.java:307: error: incompatible types: Vector<Vector> cannot be converted to Vector<Vector<Object>>
            return super.getDataVector();
`

JLLeitschuh, 2018-09-25:
Wow, this is old if it's still using the `Vector` type.

SamCarlberg, 2018-09-25:
@JLLeitschuh RobotBuilder was written in the summer of 2012 by someone more familiar with the version of Java running on the cRIOs (1.2) than with the standard library. It's also using Swing types that use `Vector` internally and overriding/using those fields (which, for some reason, are `protected` not `private`)

SamCarlberg, 2018-09-25:
The easiest solution here would be to cast the fields to the raw `Vector` type, which I believe was the original type of these fields before Oracle added the generic type parameters in 9.

Or we could do it the right way and make our classes not extend UI classes.  That would be a _lot_ of work, though.

closed 2019-01-06 by sciencewhiz