wpilibsuite/RobotBuilder #142: Nondeterministic output of files makes git merging very difficult

sataylor000, 2019-02-20:
We have noticed that Robotbuilder has some nondeterministic output that makes save files and exported code change in ways that create headaches for merging.

I lumped them into one "determinism" request, but you might decide to "divide the question".

Issue 1):  The tab/space indentation before the //BEGIN and //END pragmas tends to randomly jitter around between exports, even for files that didn't get changed in any other way.  It seems fairly random that a subset of files will suddenly have an extra tab added before such pragmas, and at other times tabs are mysteriously removed.   In many cases, this is in a file for which this was the ONLY change, and is in a subsystem or command that hasn't been modified.

Issue 2):  The saved .yaml file appears to output in completely random order of components between saves.  Most of the time, when I look at diffs (e.g., in Github Desktop), we see that the entire file shows up as modified, even if we only changed a single component (like a joystick button # mapping), or added/removed a single command/subsystem/component.  This really kills us when we are working in multiple branches because it is extremely hard to diff the changes for merging.  We have to pore over the files manually and HOPE we can manually create a new merged save file that has the right changes in the right places.  Or, as things have gotten larger as we approached Stop Build, we ultimately invoked a procedure where we locked down changes to RobotBuilder to be in only one branch at a time, eliminating it as a merge conflict.

If this .yaml file could be written out in a consistent order every time, such that a changed component only changed the lines for that component relative to the old save file, then merging would become much simpler for users.

sciencewhiz, 2019-08-08:
For Issue 1, I've seen it export with one level of indentation, and then next export changes the level of indentation, but doesn't change after that. Have you seen something different?

sataylor000, 2019-08-09:
I had to dig through hundreds of commits in GIT from our code last season,
but I did find at least one (I stopped at that point) case where a commit
indented a specific pragma outwards, and a later commit indented the SAME
pragma inwards.

So yes, it does happen, but it's pretty rare (36 different files had
indentation changes before I hit the same file/pragma)

-Scott