wpilibsuite/RobotBuilder #145: #include not parsing for 3rd party extension CPP exports

cpapplefamily, 2019-06-27:
I have successfully created a few RB extensions for Java. [Expanded TalonSRX and Spark MAX]. When I try to complete these extensions for CPP the resulting code will not build. I have tried to only use the unmodified installed TalonSRX with similar results. After manually adding the #include "ctre/Phoenix.h" to both the MySubsystem.h and MySubsystem.cpp the CPP project builds successfully. Looking at the C++ Export.yaml for the TalonSRX extension the syntax looks inline with the hard coded objects. Even deeper digging into the .../export/cpp/Subsystem.h has no parser for ID=INCLUDES, and the .../export/cpp/Subsystem-includes.cpp is failing the #if($default_command != "None")\#include "Commands/#class($default_command).h"#end

Admittedly I'm no master coder and CPP is foreign to me. I'm depending on the build button to tell me if the robot code basic objects creations are correct. I can also manually add the #include "rev/CANSparkMax.h" then my extensions works.

cpapplefamily, 2019-06-27:
Update: I have cloned the current master branch and attempted a few solutions for my self.

The `#if($default_command != "None")#include "Commands/#class($default_command).h"#end` is not failing. With further investigation  and understanding this is for the "Default_command" not objects.
Adding `${helper.getImports($subsystem, "RobotMap")}` to the end of the Subsystem-includes.cpp now created a INCLUDES field to the export and satisfies my needs. 

There is no Subsystem-includes.h export class so to quickly test my theory I added  
`#@autogenerated_code("includes", "")
#parse("${exporter-path}Subsystem-includes.cpp")
#end`
To the Subsystem.h file just under the hard-coded `\#include` . This worked only if the subsytem is new with 3rd party devices in the project but if changes are made this field is not updated.  Maybe this is a artifact of my quick test and not creating a Subsystem-includes.h. This file from what I see does not need  everything that the Subsystem-includes.cpp has.

If I'm on the correct path i can fork the master and create a pull request with the changes I have found.

sciencewhiz, 2019-12-30:
Fixed in #174

closed 2019-12-30 by sciencewhiz