wpilibsuite/RobotBuilder #146: Needs a rewrite for new Command-Based and PIDController

Daltz333, 2019-07-03:
https://github.com/wpilibsuite/allwpilib/pull/1682 and https://github.com/wpilibsuite/allwpilib/pull/1300 make major changes to the existing Command-Based and PIDController system. RobotBuilder needs to be updated to reflect this.

Nico1999, 2019-07-03:
I'm willing to take this on. I'll start working the changes in on my fork.

cpapplefamily, 2019-10-19:
I have been taking a poke at the New Command Based (Java) export of RobotBuilder. I have a few hours into it. I forked from @Nico1999 so hopfully he can or some one else can use the bits to get the CPP side done. Truthfully I have no clue to the CPP side but the Java side is comming along well. Two Questions I came across thus far:

1. When I open the project I get a warning saying This Project is not compatible with this version of the extension. Please...  I assume its something to do with a config file of some sort in the RB exporter 

2. With the Command Needing to know what Subsystem it requires as a parameter into its constructor I'm finding it difficult to parse the object Name. The current Code snip `#variable(${command.getProperty("Requires").getValue()})` returns the object name but not the Class Name that will be needed to assign the parameter type. 
`public ExampleCommand(ExampleSubsystem subsystem) {
    m_subsystem = subsystem;
    addRequirements(subsystem);`
I'm thinking about adding a parameter to the ExportDescription and a getter macro for `objectName: "#class($Short_Name)"`

I already added a parameter `NewObject: "private final .....` to create a field that will declare and construct the objects on one line like most of the examples have.

 
[image]

cpapplefamily, 2019-10-19:
> When I open the project I get a warning saying This Project is not compatible with this version of the extension. Please... I assume its something to do with a config file of some sort in the RB exporter

Fixed that in the wpilib_preferences.json `"projectYear": "Beta2020",`

cpapplefamily, 2019-10-19:
> 2\. With the Command Needing to know what Subsystem it requires as a parameter into its constructor I'm finding it difficult to parse the object Name. The current Code snip [...]

After hacking through it I found a better way so I removed my new variables objectName, objectType.

sciencewhiz, 2019-10-25:
#155 will be merged for beta 2, which will probably cause some minor conflicts in your work.

sciencewhiz, 2020-01-01:
Unfortunately, there's still too much work on porting RobotBuilder to support the New Command Framework for the 2020 season officially. However, @cpapplefamily 's work has been merged to the NewCommandFramework branch and PRs are welcome against that branch.

closed 2020-10-19 by sciencewhiz