wpilibsuite/RobotBuilder #212: Use a generated wpilib version based on git tag

sciencewhiz, 2020-04-10:
Similar to how is done for wpilibj, but should only be the release tag.

ThadHouse, 2020-10-09:
This is already done. We use the version number from the tags, and this is what gets placed in the manifest. The only fixed version number we have is the fallback if there is no manifest. 

https://github.com/wpilibsuite/RobotBuilder/blob/master/build.gradle#L128

I see no difference of using the manifest vs using generating a file, especially since for local builds the version would be incorrect for generated projects anyway.

sciencewhiz, 2020-10-11:
The value in the manifest is bad on anything other then a release build. I can't give someone a development build or a PR build, because it uses the value in the manifest which isn't a valid GradleRIO version. It also doesn't work when you use ./gradlew run. If we use a generated version file, with appropriate rules to translate development and local build versions into the correct gradlerio version, it would solve all those problems, and keep from having to update the version number each release as a fall back.

ThadHouse, 2020-10-11:
How would we know the correct GradleRio version? We could make it just generate the primary version number without any extra bits easily enough.

But that would all be done at build time, so doing the manifest vs a generated file wouldn't matter. You could do the trying to be smart logic for both methods.

sciencewhiz, 2020-10-11:
The manifest + logic works for everything except running from gradlerio

PR #605 (2024-07-29): Add action to tag on allwpilib tag dispatch
Disables export tests and updates hardcoded version prior to tagging 
Re-enables export tests after tagged
Fixes #212
commit 97752c0d (2024-07-29): Add action to tag on allwpilib tag dispatch (#605)

Disables export tests and updates hardcoded version prior to tagging
Renables export tests after tagged
Fixes #212
closed 2024-07-29 by sciencewhiz