wpilibsuite/RobotBuilder #257: commands with parameters but without parameter presets do not generate properly

sciencewhiz, 2020-10-11:


bradamiller, 2020-11-28:
If a command takes a parameter and a SmartDashboard button is automatically created for it, the parameter is missing from the new. So the DriveInches command constructor takes a number of inches, but it generated this:
    SmartDashboard.putData("DriveInches", new DriveInches( m_chassis ));
instead of:
    SmartDashboard.putData("DriveInches", new DriveInches(inches,  m_chassis ));

Adding a preset caused it to generate the correct code.

sciencewhiz, 2020-11-28:
Without having a preset, there isn't any compilable code that could be generated for the Smartdashboard call. So, the only option is to not generate the smartdashboard call for that command, or have a validation that would cause an error if it is put to SmartDashboard with a parameter without a parameter preset.

Best I can tell this has been this way ever since parameters were introduced, so it's low on my priority list.