wpilibsuite/RobotBuilder #258: Commands that are not used in joystick buttons are not exported to smartdashboard

sciencewhiz, 2020-10-11:
Broken by #214

cpapplefamily, 2020-10-19:
I have a local copy of the NewCommandFramwork branch that is not attached to any git repos. I believe I solve this and would like to get advised how to get it pushed to you.

Should I delete my fork on my git account of robotbotbuilder since I have not real idea the statuse of mine to the WPI. Then apply my changes and attempt to request a pull request.

Or just post my new Macro and code snip-it for the RobotContainer-constructors.java?

In a nutshell the reason this was broken was RobotContainer-constructors.java did not iterate through the commands to check if the box was check to create the required smartdashboard buttons, and my new macro is to create a command that adds the requires subsystem variable that the new command base model uses most of the time.

sciencewhiz, 2020-10-19:
You don't have to delete your fork, just reset the branch.

In the future, a best practice is to use feature branches to avoid issues like this.

cpapplefamily, 2020-10-19:
I admittingly don't use Git properly most due to lack of knowledge. I'll clean things up and get a pull request in soon

closed 2020-10-24 by sciencewhiz