wpilibsuite/RobotBuilder #260: PIDSubsystem's PIDController is ungrouped

sciencewhiz, 2020-10-17:
In PIDSubsystem, the PIDController is not added as a child of the subsystem.

cpapplefamily, 2020-10-18:
I have been trying to wrap my head around what I have been seeing regarding this. I hope the information I have hear is helpful. Also I believe I found a new issue that at lease is in common with this as it is related  to the PIDSubsystem More on that in the next post. I attached a screen capture
[image]

cpapplefamily, 2020-10-18:
I have been testing more with this possible bug today trying to collect more data. I also believe I uncovered a few new issues related to at least the PIDSubsystem in Teleop mode. 

 

I downloaded the latest branch of NewCommandFramework  since my last push of the work I have done last year I had continued to hack in and learn more how the program worked and thought It would be a good Idea if I worked from the current state. I programed a PIDSubsystem using a VictorSP and an AnalogPotentiometer. I observed the same results with the PIDControler not assined to the PIDSubsystem but as Ungrouped. I am not sure if this is the reason these controls don’t seem to have any effect when I try to enable and control the hardware. Picture or the the LiveWindow attached here: 
[image]

I then decided to switch the robot to teleop and select a Suffleboard button I had to manually add to the RobotContainer. **(new Bug One)** that I admittingly did not likely complete when I pushed the changes last fall.  When this command exicuted the robot program crashed **(new Bug Two)** on the motor.pidWrite(output + kF * setpoint); line in the useOutput method. I added some print statement to confirm the [output, kF, and setpoint] had values. It seems the speed controler in this case my VictorSP is not instantiated but it does work in Test_Mode so I’m stump for now here.  

Here is the stack trace when the program crashes: 

```
Set Point Command Init ﻿ 

﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ Unhandled exception: java.lang.NullPointerException ﻿﻿ frc.robot.subsystems.PIDSubsystem1.useOutput(PIDSubsystem1.java:90) ﻿﻿﻿ 

﻿﻿﻿﻿﻿﻿ Error at frc.robot.subsystems.PIDSubsystem1.useOutput(PIDSubsystem1.java:90): Unhandled exception: java.lang.NullPointerException ﻿ 

﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 1 ﻿﻿ Robots should not quit, but yours did! ﻿﻿ edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:291) ﻿﻿﻿ 

﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ The startCompetition() method (or methods called by it) should have handled the exception above. ﻿﻿ edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:293) ﻿﻿﻿ 

﻿﻿﻿﻿﻿﻿     at frc.robot.subsystems.PIDSubsystem1.useOutput(PIDSubsystem1.java:90) ﻿ 

﻿﻿﻿﻿﻿﻿     at edu.wpi.first.wpilibj2.command.PIDSubsystem.periodic(PIDSubsystem.java:47) ﻿ 

﻿﻿﻿﻿﻿﻿     at edu.wpi.first.wpilibj2.command.CommandScheduler.run(CommandScheduler.java:239) ﻿ 

﻿﻿﻿﻿﻿﻿     at frc.robot.Robot.robotPeriodic(Robot.java:59) ﻿ 

﻿﻿﻿﻿﻿﻿     at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:285) ﻿ 

﻿﻿﻿﻿﻿﻿     at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:86) ﻿ 

﻿﻿﻿﻿﻿﻿     at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:276) ﻿ 

﻿﻿﻿﻿﻿﻿     at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:348) ﻿ 

﻿﻿﻿﻿﻿﻿     at frc.robot.Main.main(Main.java:31) ﻿ 

﻿﻿﻿﻿﻿﻿  ﻿ 

﻿﻿﻿﻿﻿﻿ ﻿Warning﻿ at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:291): Robots should not quit, but yours did! ﻿ 

﻿﻿﻿﻿﻿﻿ Error at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:293): The startCompetition() method (or methods called by it) should have handled the exception above. ﻿ 

﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 0 ﻿﻿ [phoenix] Library shutdown cleanly ﻿﻿  ﻿﻿﻿ 

﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 0 ﻿﻿ [phoenix-diagnostics] Server shutdown cleanly. (dur:403|0) ﻿﻿  ﻿﻿﻿ 

﻿﻿﻿﻿﻿﻿ ********** Robot program starting ********** ﻿ 

﻿﻿﻿﻿﻿﻿ NT: server: client CONNECTED: 172.22.11.1 port 61386 ﻿ 

﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 44003 ﻿﻿ FRC:  No robot code is currently running. ﻿﻿ Driver Station ﻿﻿﻿ 

﻿﻿﻿﻿﻿﻿ Watchdog not fed within 0.020000s ﻿ 

﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 1 ﻿﻿ Loop time of 0.02s overrun 

﻿﻿ edu.wpi.first.wpilibj.IterativeRobotBase.printLoopOverrunMessage(IterativeRobotBase.java:309) ﻿﻿﻿ 

﻿﻿﻿﻿﻿﻿     SmartDashboard.updateValues(): ﻿Warning﻿ at edu.wpi.first.wpilibj.IterativeRobotBase.printLoopOverrunMessage(IterativeRobotBase.java:309)0.001881: Loop time of 0.02s overrun
 

```
I did launch the version of Robotbuilder that was included with the 2020 installer that creates the Old Command Base model an everything works correctly as expected.

sciencewhiz, 2020-10-18:
Could bug 1 be #258?

cpapplefamily, 2020-10-18:
I'll add a Joystick and report.

cpapplefamily, 2020-10-18:
I do believe I uncovered the NullPointerException. Turns out Robotbuilder is creating the hardware in two steps and not as final. Looking at the sample code in FRC-Docs has the deceleration and assignment at the top of the class. RobotBuilder Declares the object in the class but assigns it in the class constructor.

I wish I could find the conversation I had with one of you SOME TIME AGO about this move to a one step Declare and Assing from the two step where the assignment was in the constructor. At that time they agreed that the two step would be returned but is this wrong?

sciencewhiz, 2020-10-18:
bug 2 doesn't have to do with where the items are declared and initialized, but rather that the speed controller was being declared locally, which hid the member variable, so it was null when used in other methods.

cpapplefamily, 2020-10-19:
> Could bug 1 be #258?

Yes, It is related. The Object did get a button when the Joystick and button was added, But this was just a Smartdashboard copy of the Joystick operation. I did find a solution to the issue that robotbuilder does not respect the check box of the command if it is checked to add button to Smartdashboard. I will post these finding on that issue.

cpapplefamily, 2020-10-20:
> bug 2 doesn't have to do with where the items are declared and initialized, but rather that the speed controller was being declared locally, which hid the member variable, so it was null when used in other methods.

I can confirm adding either one of these lines
```
 addChild("PID Controller", getController());
//or
addChild("PID Controller", m_controller);
```
to my PIDSubsystem class constructor does add the PIDControler to the subsystem, but the LiveWindow control still has no effect when the robot is in Test mode. 

Is this because I'm putting this in my Robots Subsystem but the Issue #2784 is adding it to the PIDSubsystem.Java?

closed 2020-10-20 by PeterJohnson

sciencewhiz, 2020-10-21:
How the PID Controller is grouped doesn't have anything to do with it running or not. I opened wpilibsuite/allwpilib#2792 for the PID Controller not running in test mode.