wpilibsuite/RobotBuilder #355: Generated putData() of commands needs to add .withName()

bradamiller, 2021-01-06:
RobotBuilder generates SmartDashboard buttons for created commands using code like this:
    SmartDashboard.putData("Forward: OneFoot", new Forward(12, m_chassis));

The .withName() decorator should be added so that it's displayed correctly in glass and shuffleboard like this:
    SmartDashboard.putData((CommandBase) (new Turn(m_romiDrivetrain, 90).withName("Turn 90")));

sciencewhiz, 2021-01-07:
Here's what I get for this line in smartdashboard and shuffleboard:
`SmartDashboard.putData("Set Wrist Setpoint: Horizontal", new SetWristSetpoint(0, m_wrist));`

SmartDashboard:
[image]
Shuffleboard:
[image]

I think Glass would show the name if there was room in the window

[image]

bradamiller, 2021-01-07:
It is using the command name and not the label. The problem I had was that I had two turn commands on the dashboard, one was called "turn 90" and the other was "turn -90", but they both displayed ambiguously as "Turn" because glass used the command name and not the putData first parameter.

sciencewhiz, 2021-01-08:
When wpilibsuite/allwpilib#3060 is resolved, then the Glass display would be similar to the Shuffleboard display

sciencewhiz, 2021-01-21:
In Glass 2021.2.1, the key is shown first, obviating this issue.

sciencewhiz, 2021-11-23:
Closing since glass now shows the name

closed 2021-11-23 by sciencewhiz