wpilibsuite/RobotBuilder #363: Instant commands have problems

jmullins16, 2021-01-19:
Defined instant command with no required subsystem. Generated code attempts to inject a subsystem parameter with the name “None” (the selection shown when no subsystems have been selected). Since there was no such subsystem (or header files), it would not build.

We created a “dummy” subsystem to assign to commands that did not need to require any of our other subsystems. This allowed it compile.

Added a parameter to the same instant command, and the autogenerated constructor now only shows the parameter, not the subsystem injected. Header file is correct.

```
    // BEGIN AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=CONSTRUCTOR
    explicit ClimberDeploy(bool climberDeploy, Dummy* m_dummy);

    // END AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=CONSTRUCTOR
```

```
// BEGIN AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=CONSTRUCTOR

ClimberDeploy::ClimberDeploy(bool climberDeploy): InstantCommand() {
m_climberDeploy = climberDeploy;
    // Use AddRequirements() here to declare subsystem dependencies
    // eg. AddRequirements(Robot::chassis.get());
    SetName("ClimberDeploy");
    AddRequirements(m_dummy);

    // END AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=CONSTRUCTOR
```

Also noticed that the generated SmartDashboard buttons do not update when adding/deleting parameters to an instant command.

So we are switching back to regular commands which seem to work fine.

sciencewhiz, 2021-01-28:
For the problem about instant commands not updating parameters, did you update the parameter presets after updating the parameters? There shouldn't be anything different between Instant Commands and Commands when generating SmartDashboard buttons.

closed 2021-01-29 by sciencewhiz

jmullins16, 2021-01-29:
It took me a bit to repro, but here's what I did.

1) Create a new project with a single subsystem and one instant command (no parameters), build no errors
2) Added a parameter to the instant command, build failed with instant command Constructor missing subsystem
3) Added a preset to the instant command, build failed with instant command Constructor missing subsystem
4) Added a new command (no parameters), build with no new errors
5) Added a parameter to the command, SmartDashboard PutData for command did not add parameter, build with error
6) Added a preset and it compiled with no errors
7) Deleted preset and SmartDashboard again missing parameter, build with errors

So I think it's in both.

sciencewhiz, 2021-01-30:
Yes, Smartdashboard failing without parameter presets is known (#257) and isn't anything fixable, other then potentially throwing a validation error.

PR #366 (2021-01-29): Fix InstantCommand when no subsystem
Replace InstantCommand-constructor with the generic Command-constructor
Don't generate subsystem include if no subsystem required
Add export tests for InstantCommand
Fix AddRequirements comment

Partially fixes #363