wpilibsuite/RobotBuilder #367: Commands do not support subsystem injection unless required

jmullins16, 2021-01-25:
Commands can only be created with an injected subsystem if the the Requires selection is made. We have some commands that need to reference a subsystem without causing a different command to be interrupted. For instance, we include the shifter pneumatic within the drivetrain subsystem, but we don't want shifting to interrupt the command controlling the drive motors. If our model is wrong, please advise.

sciencewhiz, 2021-01-26:
unfortunately, this isn't very straightforward to change without affecting templates which makes any fixes not backwards compatible. Normally we would try to avoid doing that mid season. 

You use c++, right? 

There's a relatively simple workaround in Java, deleting the addrequirements and auto generated code block so it isn't regenerated. 

The C++ template is not as fine grained. You could still do that, but it would affect the constructor to, so it wouldn't update if you added parameters, for example.

jmullins16, 2021-01-27:
I kind thought this would be a big change. Almost seems like there needs to be a "Uses" as well as "Requires".

Yes, we are a C++ team and unfortunately we were blocked by the CTRE extensions during beta, so didn't get this far.

We will probably do as you say and use RB to generate the initial patterns and then take out the auto-generated markers for now. Still like to use the tool to let the students focus on the implementation more than the overall structure.

sciencewhiz, 2021-01-27:
One change we discussed is making the subsystems public in robotcontainer. This would make them accessible in the commands and would not require a big template change.

PR #370 (2021-01-30): Make subsystems public in RobotContainer
Allows commands that don't require the subystem to access them (Fixes #367), This also allows commands to require multiple subystems and avoids huge dependency chains for CommandGroups.
commit 290357c8 (2021-01-30): Make subsystems public in RobotContainer (#370)

Allows commands that don't require the subystem to access them (Fixes
 #367), allows subystems to require multiple subsystems manually and
 avoids huge dependency chains for CommandGroups.
closed 2021-01-30 by sciencewhiz