wpilibsuite/RobotBuilder #381: C++ Removing a required subsystem overwrites the entire command file

jmullins16, 2021-02-02:
We removed a required subsystem from a command, and the entire command file was changed back to the initial template--not just the constructor.

sciencewhiz, 2021-02-02:
I can't duplicate this

jmullins16, 2021-02-02:
Maybe our YAML file has a problem.

https://github.com/usfirst-frc-2135/software/blob/dev/2021/Robot2021.yaml

In order to export this, it will require adding the Phoenix library and commenting out the Xbox button hooks in RobotContainer. (Build)

Add a line of code to ClimberRun in the Initialize method. (Build)

Change ClimberRun Requires setting to None.

Code is missing. Hope this helps.

sciencewhiz, 2021-02-03:
I was able to reproduce it with your file. The issue is that Robotbuilder is picking up the parameter as a type, but only when there isn't a required subystem. So it think the type is changing which causes the overwrite.

PR #387 (2021-03-27): Fix C++ Type detection
Fixes #381
closed 2021-03-27 by sciencewhiz