wpilibsuite/RobotBuilder #383: C++ RobotBuilder adds extra lines VS Code removes whitespace

jmullins16, 2021-02-04:
An low priority inconvenience, but wanted to document it.

When using an IDE that removes whitespace, some of the lines that RobotBuilder segments are changed. When RobotBuilder runs again, it updates because the existing snippet no longer matches what RB wants to insert. Specifically, it looks like RB puts CRLF and a single space on some lines in this segment. VS Code trims the whitespace (at least ours does) when that file is edited, and then RB puts it in again on the next pass. It generates a lot of confusing changes in git for the students whenever they touch a file.

It would be nice if RB did not ever leave trailing whitespace, and even better if it never generated code with more than a single blank line between statements or BEGIN/END guards.

```
    // BEGIN AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=DECLARATIONS
    SetSubsystem("Drivetrain");

 
 

 
 

 AddChild("Diff Drive", &m_diffDrive);
 m_diffDrive.SetSafetyEnabled(true);
m_diffDrive.SetExpiration(0.1);
m_diffDrive.SetMaxOutput(1.0);

 
 

 
 

 AddChild("Shifter", &m_shifter);
 

 
 

    // END AUTOGENERATED CODE, SOURCE=ROBOTBUILDER ID=DECLARATIONS
```

cpapplefamily, 2022-04-15:
I have struggled to get the code templates in RB to output good indented white space formatted code. Its not only the C++, the Java is just as bad. If the other developers have any tips on this I just might take this on. Are there hidden control characters or ascii codes that can be inserted? DM me and we can chat.

PeterJohnson, 2022-04-15:
It would be a ton of work, but we would likely need to switch to a different template language with better control over whitespace to fix this 100%.  I personally like Jinjava (a version of Jinja in native Java), https://github.com/HubSpot/jinjava

cpapplefamily, 2022-04-15:
since posting I have looked at this some and have a few ideas working. I was inspired by the Autogeneratedcode macro that takes in a whitespace variable #lineComment macro that you can pass in the spaces and comments. I have also started to create a few copies of some macros like #variable to #variabletabbed and #constructor to #constructorTabbed that takes in a $space parameter to pad the output.

It didn't seem to allow me to overload the macro so new macros were created for testing but in the end when I'm happy I could reduce the new macros back to the originals to attempt to find all instance and do some project wide find/replace.  

The work can be found here if interested.  [https://github.com/cpapplefamily/RobotBuilder/tree/whiteSpace](https://github.com/cpapplefamily/RobotBuilder/tree/whiteSpace)

cpapplefamily, 2022-05-30:
@sciencewhiz I see you have been busy approving pull request and preparing for the next round of updates. I have been sitting on this one for some time. White space control has been a real struggle. I have this "whiteSpace" branch that has macros that take a string to pre-load constructors, functions, and line Comments with required white space.

it is sort of a hack but would get the resulting output closer to well indented code. 

Should I put some more time into this?

sciencewhiz, 2022-06-11:
I haven't had a chance to take a look at it yet

sciencewhiz, 2022-09-04:
@cpapplefamily rather then creating copies of macros, I do think it would be better to add the whitespace argument to the existing macro.

cpapplefamily, 2022-09-04:
@sciencewhiz My Macro Copy was to avoid breaking the application and more about prototyping concepts. 

Its been some time but I peeked at trying to use a Formatter command but recall having issues. What they were is lost to me because of time. Issues could have been my lack of knowledge on how to use or other conditions.