wpilibsuite/RobotBuilder #472: Robot Builders automated SmartDashboard buttons

cpapplefamily, 2022-02-28:
I'm curious if we should add an option to disable the Smartdashboard.putdata for each joystickbutton added via the Robotbuilder Operator Interface.

I had a few teams over the weekend with basic code running that was developed using robotbuilder that has seen alot of Differential Drive not updated enough logs. With the log I have seen the buttons.java runtimes of 0.0246+ms. I'm making a leap thinking the duplicated add command to Smartdashboard (Selectable) and the button added with each joystickbutton is overloading the network table. Please confirm that possibility or defunct it.

sciencewhiz, 2022-02-28:
The fact that a command is put to SmartDashboard should have no impact on the time it takes to poll the buttons. The code only polls the button state: https://github.com/wpilibsuite/allwpilib/blob/dae18308c9aecc041cebb5dc45e32fc508842e9b/wpilibNewCommands/src/main/java/edu/wpi/first/wpilibj2/command/CommandScheduler.java#L257-261

Overruns from too much stuff on network tables would show up here: https://github.com/wpilibsuite/allwpilib/blob/dae18308c9aecc041cebb5dc45e32fc508842e9b/wpilibj/src/main/java/edu/wpi/first/wpilibj/IterativeRobotBase.java#L331-336

You'll only get differential drive not updated enough if there's 100ms of delay between calls. So even if there is 20.46ms of delay in reading buttons, there's still almost 80ms of other stuff delaying.

With that being said, there probably should be an option to not put the Joystick Button commands to SmartDashboard, like there is for other ways to create commands.

cpapplefamily, 2022-02-28:
Thanks for the clarification. There were a list of all the Subsystem Periodic() methods that all seemed naggable. I just stopped at that one since it seemed a bit egregious.

Anyhow thanks for agreeing that there should be a way to disable additional buttons on the overcrowded dashboard.  Put that in the features request.

cpapplefamily, 2022-04-15:
#473

sciencewhiz, 2022-05-29:
Fixed by #473

closed 2022-05-29 by sciencewhiz