wpilibsuite/RobotBuilder #508: Discussion: Extensions vs Hardcoded for WPI components.

cpapplefamily, 2022-11-30:
I was panning through the 2023 changes through the Beta links and the PS4Controller class jumped out at me. Since this is a WPILib class I though about adding it to Robotbuilder like the Xbox Controller. I already created a working Extension and thought would it be possible to include an Extensions folder in the repository to easily add new components as needed vs editing the individual files that make RobotBuilder operate. 

Maybe there are other reasons to keep all of the WPILib components coded to the few internal files that I'm not aware of. If that is the case I can move my edits into the necessary files and create a Pull Request.

cpapplefamily, 2022-11-30:
I believe I may have found that limitation. Seems that an extension does not "Currently" have knowledge of the macros with in the PalletteDescription.yanl file #Button("") 

```
!Component
  name: PS4 Button
  type: PS4 Button
  help: An PS4 Controller button that binds a driver action to a command.
  properties:
    - !ParentProperty
      name: Joystick
    - !ChoicesProperty
      name: Button
      choices: ["Circle","Cross","L1","L2","L3","Options","PS","R1","R2","R3","Share","Square","Touchpad","Triangle"]
      default: "Circle"
    #Button("")            <---------------- This Here
    - !BooleanProperty
      name: Send to SmartDashboard
      default: false
```

I suppose the Extension could recreate the required code but that violates -DRY-

sciencewhiz, 2022-12-04:
What is the advantage of doing everything as extensions?

cpapplefamily, 2022-12-04:
I'm not sure there is or is not an advantage. Possibly more people would develop components if they had templates as examples. Sure most devices we use fall into a hard-coded. 
One advantage could be like with the PS4controller that is a WPILib class could be added to the application without waiting for a revision release. Users that want it can simply get the files from the git repo and extract to the appropriate location. 
Questions are there more components that like the joystick family of devices that could all use the Joystick class but gain some useful choices when using the XboxController class or PS4Controller class?

If users have an extention that would fit inclusion to the project it would be easier for them to submit them to a central extention repo verses cloning the project and inserting the bits into the various locations.

I point every team and student to a repository of extention I created that are mostly 3rd party enhanced versions of their existing extentions if they have one. Navx and Rev currently don't. I do.